Sensirion/arduino-i2c-sen5x #1: Not enough space in buffer

IcingTomato, 2022-01-17:
Hello, I am a Application Engineer in Seeed Studio.
[image]
I have this problem in Arduino Uno.
Not enough space in buffer.

psachs, 2022-01-17:
Hi,

thanks for the report. I have an Arduino Uno R3 and the example works for me.  I tested the Arduino example with multiple Arduino boards and ESP32 without any problems.
I am using the following setup:
* Arduino IDE 1.8.13
* Sensirion Core 0.5.3

Does the issue persist if you reset the board? Resp do you always have this issue or just sometimes?

Can you check if you have a stable power supply and i2c connection (e.g. no loose cables, bad contacts).

I will try to reproduce the error. Since you get I2C bus errors I suspect the "not enough space in buffer" error to be an unwanted side effect of the I2C error. This error should only occur if the specified buffers in the driver are not setup right.

Best regards,
Pascal

IcingTomato, 2022-01-21:
Fine.
Thank you very much. The cable problem.

closed 2022-01-21 by IcingTomato