Sensirion/arduino-i2c-sen5x #4: NOx set state

yukimach, 2023-01-02:
Hi, 

I'm just wrapping my head around how setVocTuningParameters and setVocAlgorithmState work and differ.

Does the algorithm state store historical data such as 12hour history, and so it differs from just storing setting parameters for VocTuningParameters - such as offsets? 

Is there a reason there is a setNoxTuningParameters but no setNoxAlgoirthmState, whilst VOC has this? I'm looking at the SGP41 repo, and it seems this can exist and can be made available? I'm trying to understand if this is possible on the SEN51. Is there more documentation on this? 

Thanks!!

tyeth, 2023-01-08:
There was something mentioned about tuning in this ticket on the gas-algorithm repo, https://github.com/Sensirion/gas-index-algorithm/issues/1
I've been experiencing a 1.0 returned float for over 24hrs. There were two minor deviations to 0.83 and 0.7 but possibly at init for one of them. I've been seeing the same in arduino sketch. Having stuck my prototype under a saucepan and using a yellow-flamed kitchen lighter for about 3seconds up under one edge and closing the pan down caused the NOx to wake up, and now all seems more realistically as described in the datasheets where an index of 100 is expected for both VOC and NOx.
It would be lovely to have a more thorough getting your design correct guide, as the temperature compensation guide is a huge part of that, presumably the voc/nox tuning is too, rather than being a fire and forget event sensing index. Or maybe I'm being overly harsh, as with the saucepan I've seen it does at least react when submitted to enough of a NOx event.

psachs, 2023-01-10:
the algorithm tuning parameters `setVocTuningParameters` can be used to specify the behavior of the gas index algorithm. E.g. the `index_offset` parameter specifies what VOC index is used under average condition (the environment condition the sensor sees most often)
Please check to full documentation in the gas index algorithm header file:
https://github.com/Sensirion/gas-index-algorithm/blob/master/sensirion_gas_index_algorithm/sensirion_gas_index_algorithm.h#L220

The state `setVocAlgorithmState` can be used to faster recover from device restarts. After at least 3 hours of operation the state can be used for short (max 10min) interruptions. The state has to be saved regular (e.g. once per minute) and set on startup **after** `init` or `reset`. The algorithm will then skip the initial learning phase and be much faster fully usable again.

the algorithm state and some of the tuning parameters are only available for VOC and not for Nox. The algorithm state does not work for the Nox signal and is therefore only available for VOC.

Please find all available documentation on the SGP41 product page:
https://developer.sensirion.com/sensirion-products/sgp41-voc-nox-sensor/

closed 2023-01-18 by yukimach