Sensirion/arduino-i2c-sen5x #5: Curiosities - Could SEN50 be made to return temp/humidity in all methods that return them?

tyeth, 2023-01-13:
Hey, I've got all three models, and using the measureValues as floats works nicely for the 55+54, but the 50 returns failed temp+humidity values (NAN) and valid particle counts (and expected NANs for nox/voc).
 I do know that there's a separate method for the sen50, but could the other method be made to not fail, as that would make most peoples lives easier. Don't drop the method for the 50, but make the other methods work as far as possible for the 50.
Thanks ❤️

Also on the curiosities front, I just got 3 new (one of each) sensors, and previously just had a single SEN55 from the same supplier. I've found them to be vibrating more than expected (fan vibrations). Particularly when held by both sides between thumb and forefinger, from next to cable to edge of QRcode. Is this expected to be quite a lot, and how would one even measure such a thing?
Is there a batch number or anything in the QRcode that tells me which fan I've got if you've used more than one?

tyeth, 2023-01-13:
I'm a muppet, it turns out I think (not mentioned on product page) that there is no temp/humidity on the SEN50. Woopsie.

Still curious about the vibration question 🤷

psachs, 2023-01-18:
Hi,
The vibration could be caused by the fan auto cleaning where the fan runs on the highest speed. The difference between the SEN50 and SEN55 is only the sensors available in the module.
Both use the same particulate matter sensor. Of course it is possible that the fan vibration is not exactly the same for each module. As far as I am aware of, the same fan should be used in all modules. Since we don't produce them on our own there might be some variation between batches within the specs.

tyeth, 2023-01-18:
Thanks that's confirmed my own thoughts. I assume they don't do a clean on first measure though (thought it was a day or 7 away unless changed).

closed 2023-01-18 by tyeth