Sensirion/arduino-i2c-sen5x #6: Sen55 auto-cleaning

Cam1729, 2023-01-13:
Hello, I am new to working with the Sensirion devices and am unsure if this is the right place to ask but I currently have a Sen55 connected to an ESP32 Dev Kit and I am trying to change the auto clean of the fan by using the sen5x.setFanAutoCleaning() function. Everything compiles fine but when I upload the code I get some kind of guru error. Its not the biggest issue since I know the fan is automatically set to auto clean every week but as someone new to this I am just curious what could be causing this. I even tried to use sen5x.StartFanCleaning(); to force it to clean but I get the same error. Thank you for your help!

psachs, 2023-01-18:
Thanks for the report. We will have a look at the code to see if this issue is reproducible and how it can be fixed. 

Which exact version of the ESP32 Dev Kit are you using and does this only happen if you use the auto clean interface or also with the unmodified example?

psachs, 2023-12-15:
I can't reproduce this issue with an ESP32 dev kit. It looks like your settings for the ESP32 might be wrong and the crash is not related to the driver

closed 2023-12-15 by psachs