Sensirion/arduino-i2c-sen5x #7: Not getting raw voc value using readMeasuredRawValues() function

Akshul-pirhoalpha, 2023-01-19:
Getting 65535 repeatedly as raw value, while I am getting voc index value. But I actaully need voc raw value in ppb. Please help.

LeonieFierz, 2023-01-20:
Thank you for your request.

The sensor does not provide VOC values in ppb. The raw value is an internal unit-less measurement tick, which is proportional to the logarithm of the resistance of the sensing element. You can find more information about the VOC Index on https://sensirion.com/media/documents/02232963/6294E043/Info_Note_VOC_Index.pdf.

How to convert form the sensor measurement ticks to the physical values for temperature and other measured values is documented in the header of the read functions or the datasheet https://sensirion.com/media/documents/6791EFA0/62A1F68F/Sensirion_Datasheet_Environmental_Node_SEN5x.pdf in chapter 6 as Scale Factor.

Regarding that you get the invalid value when using the readMeasuredRawValues, we will look into it. Please note that for the first second after you started the measurement the values are invalid. Which sensor of the SEN5x family do you use? What value do you read for VOC Index?

mdede439, 2023-01-24:
I am interested in the raw values as well.  I thought the SEN55 had a SGP41 inside to do this kind of measurement.  There is no way for the SEN55 to output the values from this sensor?

LeonieFierz, 2023-01-25:
Hi @mdede439 
We tested out to read raw values with readMeasuredRawValues(...) to get VOC/NOx raw values from SEN54/55, which is working for us on a ESP32. Can you send us the code snippet which is not working for you please?

mdede439, 2023-01-25:
That is my problem, I do not know how.  I tried to use the example Arduino sketch for the SEN5x but replace the  
"error = sen5x.readMeasuredValues()" function with one I found in the library for this device ,"sen5x.readMeasuredRawValues()" but it did not work.  I got an error related to the a mismatch of the variable types.  I am still unclear as to what is wrong.  If you have a way of getting those values out of the sensor, the NOx and VOC values and not the index values, then I would truely appreciate learning how to do it!  When I read that the actual values were not available I gave up.

LeonieFierz, 2023-01-26:
Hi @mdede439 

You can try it out with the following code snippet

``` c++  
    int16_t rawHumidity;
    int16_t rawTemperature;
    uint16_t rawVoc;
    uint16_t rawNox;
    
    error = sen5x.readMeasuredRawValues(
        rawHumidity, rawTemperature, rawVoc,rawNox);

    if (error) {
        Serial.print("Error trying to execute "
                     "readMeasuredRawValues(): ");
        errorToString(error, errorMessage, 256);
        Serial.println(errorMessage);
    } else {
        Serial.print("Raw Humidity:");
        Serial.print(rawHumidity);
        Serial.print("\t");
        Serial.print("Raw Temperature:");
        Serial.print(rawTemperature);
        Serial.print("\t");
        Serial.print("Raw VOC:");
        Serial.print(rawVoc);
        Serial.print("\t");
        Serial.print("Raw NOx:");
        Serial.println(rawNox);
    }

```

mdede439, 2023-01-26:
Thanks so much, that did work.  I tried "almost  "exactly what you posted except I used int16_t for all four variables and it would not compile.  The errors I got did not point me in to the direction of having an incorrect data type. Now I just need to understand what the data is telling me.  I can see why having an index value makes it easier to understand.  
However, I do see the rawHumidity and rawTemperature values wildly off from that I would expect.  Actual room temperature is around 23.5 C and rawTemperature is around 5695 from the sensor.  Actual room humidity is around 38% and rawHumidity from the sensor is around 2307.  The numbers do not make sense.  Am I to assume there must be some post processing to this data to get usable data?

LeonieFierz, 2023-01-27:
@mdede439 please scale the raw values with 200 for the temperature and 100 for the humidity, as described in the header of the driver https://github.com/Sensirion/arduino-i2c-sen5x/blob/master/src/SensirionI2CSen5x.h

LeonieFierz, 2023-01-27:
@Akshul-pirhoalpha 
As you see in the messages above we could read the raw values for VOC with SEN55 and SEN54.

mdede439, 2023-01-28:
Yes, Thank you for your help!  Really appreciate it.

closed 2023-02-17 by qfisch