Sensirion/arduino-i2c-sen5x #8: Code documentation inconsistent with PDF - startMeasurementWithoutPm()

arc12, 2023-04-24:
The code documentation for startMeasurementWithoutPm() states:
>      * This command is only available in idle mode. If the device is already
>      * in any measure mode, this command has no effect.

The application note "Reduced Power Operation for SEN5x - Application note describing SEN5x reduced power mode" v1.0 shows a state diagram with transitions DIRECTLY between the gas-only and PM-inclusive modes.

Could you please advise which is correct (and fix the incorrect!)

arc12, 2023-04-24:
Further investigation:
the "embedded" and RPi repositories say the same as above whereas the Python indicates a direct mode transition can be used for firmware versions greater than 1.0.
Consulting the SEN5x datasheet "Version 2 – D1 – March 2022" suggests direct mode change is OK for firmware version at least 2.0.

It looks like a good "tidy up" is needed!

sdmueller, 2023-04-25:
Hi @arc12

You are right, direct mode change is OK starting from firmware 2.0.
The note in the Python repository is correct and the comment in the other repositories is wrong.
We will correct that as soon as possible.

Thanks for pointing out this issue!

sdmueller, 2023-04-26:
The documentation for the startMeasurement command is now correct in all the SEN5X repositories.

closed 2023-04-26 by sdmueller