Sensirion/arduino-i2c-sen5x #10: same valu of PM10 PM2.5 PM4

CDNNDR, 2023-09-02:
Hello, 
I'm using a SEN54 and SEN55 with an ESP32 C3 board in deep sleep mode. I run the code and I let SEN54 run for 20 seconds before stop measuring and enter the board into sleep. I left the board running for days but I always get the same value of PM10 PM2.5 PM4, it's pretty weired.

what I'm missing? I used the code example available in this rep.

best

psachs, 2023-09-05:
Please note that the definition of the PM values is <=.
E.g. PM2.5 is all particles smaller than 2.5um, and PM10 is all particles smaller than 10um.

So e.g. the PM10 includes all smaller particles. If the values for all signals are the same or similar, than you only measure very small particles.

If you want to test the sensor functionality you can e.g. use match near the sensor to create some particles. It should also work by ripping apart a handkerchief near the sensor.

qfisch, 2023-12-15:
I will close taht issue since we did not get more information. Feel free to re-open it if you require more support

closed 2023-12-15 by qfisch

rajeshdoshi, 2024-04-16:
Does that mean PM10 value = read PM10 - PM2.5 value??

qfisch, 2024-04-16:
Hello @rajeshdoshi 

It means that the PM10 value includes the particles measured by the PM4 value (and PM2.5 by extension).  
By definition: PM10 ⊇ PM4 ⊇ PM2.5

I hope it is clear
Q

rajeshdoshi, 2024-04-16:
So that means
  Pm25=Pm25-Pm1;
  Pm4=Pm4-Pm1-Pm25;
  Pm10=Pm10-Pm1-Pm25-Pm4;
 Am I Correct?

qfisch, 2024-04-16:
well, I am not sure what you want to express with the minus there...
if it is a minus operator, then no.
since Pm25 = Pm25 - Pm1 can not be correct unless PM1 is 0

rajeshdoshi, 2024-04-16:
Then how do we calculate PM2.5  PM10 values? 
So basically, values calculated by a minus operator will give value from say PM1 to PM2.5, PM2.5 to PM4 and so on.
Am I being correct here?

qfisch, 2024-04-16:
PM10 = number of particles smaller than 10µm (in µg/m³)
PM4 = number of particles smaller than 4µm (in µg/m³)
PM2.5 = number of particles smaller than 2.5µm (in µg/m³)
PM1 = number of particles smaller than 1µm (in µg/m³)

StmLord, 2025-02-11:
I have the same issue, PM2.5, PM4 and PM10 ALWAYS have the same value. I have compare it with other sensors from nova and Honeywell and all other sensors are showing different values from PM10 to PM2.5.

qfisch, 2025-02-11:
Hello @StmLord 
I will need a bit more details here...
What are your testing condition were you obtain ALWAYS the same values ?
What are your the reference of your sensors from Honeywell and Nova ?