Sensirion/arduino-i2c-sen5x #11: setVocAlgorithmState Example

protttoy, 2023-11-09:
I am trying to get the voc algorithm state and set it upon startup.

For getting the algorithm array
```
byte statesize = 12;
byte state[statesize];
int error = sen5x.getVocAlgorithmState(state,statesize);
```

> Output : { 49, 229, 238, 62, 0, 49, 202, 99, 185, 252, 63, 123 }

After that when saving the state, "**Not enough space in buffer**" error triggers for the following code:

```
byte statesize = 12;
byte state[statesize] = { 49, 229, 238, 62, 0, 49, 202, 99, 185, 252, 63, 123 };
char errorMessage[256];
int error = sen5x.setVocAlgorithmState(state,statesize);
if(error)
{
    errorToString(error, errorMessage, 256);
    Serial.println(errorMessage);
}
```

The error occurs from the `Sension Arduino Core --> SensirionI2CTxFrame.cpp --> uint16_t SensirionI2CTxFrame::_addByte(uint8_t data)` function. 

SEN5x datasheet says in section 6.1.11 that there is Checksum returned when reading. Do I not send the checksum when using the setVocAlgorithmState ? An example of how to set the state would be really appreciated. Thank you

Rol-la, 2023-11-14:
On what hardware are you working? 
On some Arduinos the i2c communication buffer is not large enough to transmit the fully assembled data frame. 
The checksum computation is implemented in the Sensirion Software library and you should not need to deal with it.

Rol-la, 2023-11-15:
Hello,
It is as you write: The VocAlgorithmState must have exactly 8 bytes (https://sensirion.com/media/documents/6791EFA0/62A1F68F/Sensirion_Datasheet_Environmental_Node_SEN5x.pdf).
A checksum is inserted every third byte => byte [0,1] is data, byte[2] is checksum, byte [3,4] is data and byte [5] is checksum ..
So you code snippet with 8 bytes in the state variable is correct.

protttoy, 2023-11-16:
@Rol-la Thanks a lot !!

closed 2023-11-16 by protttoy