Sensirion/arduino-i2c-sen5x #15: ESP32-S3-DevKitC-1 v1.1 Compatibility with Sensirion I2C SEN5X Arduino Library

daniilnahl, 2025-06-16:
Hello Sensirion Team,

I am working with an ESP32-S3-DevKitC-1 v1.1 board and plan to integrate a Sensirion SEN54 SDN-T air quality sensor via I2C.

I've been exploring your driver repositories and found two main options:

- The arduino-i2c-sen5x Arduino library.
- The embedded-i2c-sen5x generic embedded C driver.

The Arduino library (the arduino-i2c-sen5x one) is the only one that appears in the Arduino IDE's Library Manager for the SEN5x family. Looking at its listed compatible architectures(on Arduino page), it primarily mentions SAMD, MegaAVR, AVR, and Mbed, but not explicitly ESP32-S3's architecture. 

Could you please confirm if the Sensirion I2C SEN5X Arduino library is **compatible** with the ESP32-S3 (specifically the DevKitC-1 v1.1)? If so, are there any particular considerations or common pitfalls I should be aware of regarding pin assignments (SDA/SCL) or specific configurations for the ESP32-S3?

If the Arduino library is not directly compatible or recommended for the ESP32-S3, is embedded-i2c-sen5x generic driver the only compatible one, requiring me to implement the sensirion_i2c_hal.c functions using the ESP32 Arduino Core's Wire library?

Thank you for your time and clarification!

LeonieFierz, 2025-06-18:
Hello @daniilnahl 

The arduino-i2c-sen5x library is compatible with the ESP32-S3 board. For the wiring, you have too look up the SDA and SCL pins specific to your board (default for ESP32-S3 DevKit C is GPIO 8 (SDA), GPIO 9 (SCL), see I2C section on https://randomnerdtutorials.com/esp32-s3-devkitc-pinout-guide/). Other then that it should be straight-forward to get the example running. Let us know if you have any troubles. 

Best, Léonie

closed 2025-06-18 by LeonieFierz

daniilnahl, 2025-07-10:
Thank you! It all worked out perfectly.