Sensirion/arduino-i2c-sen5x #16: How to operate the VOC sensor and save power

oscgonfer, 2025-08-08:
We are using the SEN5X series and we would like to know how it should be operated if we want to have "interrupted" operation to save power. Ideally we want:

1. Turn on measurement mode for 30s
2. Get readings after 30s
3. Stop measurement mode for 150s
4. Go back to (1)

The question goes onto how we should interact with VOC state algorithm. The datasheet says:

[image]

Does this mean that we could do:

1. (If available, feed state of VOC state algorithm)
2. Turn on measurement mode for 30s
3. Get readings after 30s
4. Get VOC state algorithm
5. Stop measurement mode for 150s
6. Go back to (1)

In this case, the VOC state is always `[0, 0, 0, 0, 0, 0, 0, 0]`, so there is something wrong. Should it be done in a different way? For instance, not stopping measurement mode if we want to measure VOCs? In that case, how can we save battery?

qfisch, 2025-08-12:
Hej @oscgonfer 
You are on the right track. The state you are reading is weird, how long is the algorithm running before reading ?

Cheers
Q

oscgonfer, 2025-08-12:
Hi @qfisch,

Thank you for the response. It's not running for much time (probably seconds?) before entering in that sequence from above forever. I left the device running in that mode for a while (hours) and the voc algorithm state is the same.

qfisch, 2025-08-13:
Hej @oscgonfer 

I will try to reproduce on my side this week and hopefully come back to you with a working code sample.
Cheers
Q

oscgonfer, 2025-08-27:
Hi @qfisch 

Indeed, I tested the approach and I get the following (I added some decoration and only print the `non-CRC` parts of the buffer):

```
[0, 28, 177, 2, 21, 88, 0, 0]
... 
[0, 45, 98, 2, 112, 191, 0, 0]
```

This works when not stopping the sensor. But if I wanted to send `0x0104` (stop measurement, assuming that's the one the datasheet refers to) between reads to save battery, what should be the process?